nordicsemi/Android-nRF-Blinky #1: can not scan devices with samsung galaxy s5 neo with android 5.1.1

schoensn, 2016-03-10:
hi 
i have a problem using the blinky application with my new mobile phone. 

if i want to scan bluetooth peripherals with the blinky app no devices are found.

the master control panel app works fine. 

when i use android studio, the android monitor shows this log: 

`BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6`

with other devices sony xperia z4 compact the app works.

roshanrajaratnam, 2016-03-10:
Hi,

Sorry for any inconveniences caused, I will have to debug this and see
what's going on. I'll look at this first thing in the morning tomorrow and
get back to you!

Regards
Roshan

hi
i have a problem using the blinky application with my new mobile phone.

if i want to scan bluetooth peripherals with the blinky app no devices are
found.

the master control panel app works fine.

when i use android studio, the android monitor shows this log:

BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6

with other devices sony xperia z4 compact the app works.

schoensn, 2016-03-11:
many thanks in advanced. 
Regards Stefan

roshanrajaratnam, 2016-03-11:
Hi Stefan,

after spending many hours of debugging the app seemed to be working fine with regards to scanning on our end. Could you try turning off/on your bluetooth please? I have tried with 3 different devices eg. 2 S6 devices and a Nexus 4 with 5.1.1 running and  they all seem to scan the device. Just make sure your dev kit is not connected to anything as well. Let me know if you have any luck.
Regards,
Roshan

schoensn, 2016-03-14:
Hi Roshan, 

I have tried to turning on/off Bluetooth, but unfortunally this does not solve my problem... and it seems that i have an additonal issue. when i start the app with ble off i get an exception: 

`03-14 09:58:59.901 4637-4650/? D/BluetoothAdapter: onBluetoothStateChange: Bluetooth is on
03-14 09:59:01.951 4637-4655/? D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6
03-14 09:59:01.981 4637-4637/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                 Process: no.nordicsemi.android.nrfblinky, PID: 4637
                                                 java.lang.RuntimeException: Failure delivering result ResultInfo{who=null, request=0, result=-1, data=null} to activity {no.nordicsemi.android.nrfblinky/no.nordicsemi.android.blinky.MainActivity}: java.lang.IllegalArgumentException: scanner already started with given callback
                                                     at android.app.ActivityThread.deliverResults(ActivityThread.java:4687)
                                                     at android.app.ActivityThread.handleSendResult(ActivityThread.java:4730)
                                                     at android.app.ActivityThread.access$1500(ActivityThread.java:197)
                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1705)
                                                     at android.os.Handler.dispatchMessage(Handler.java:102)
                                                     at android.os.Looper.loop(Looper.java:145)
                                                     at android.app.ActivityThread.main(ActivityThread.java:6873)
                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                     at java.lang.reflect.Method.invoke(Method.java:372)
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404)
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199)
                                                  Caused by: java.lang.IllegalArgumentException: scanner already started with given callback
                                                     at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerImplLollipop.startScanInternal(BluetoothLeScannerImplLollipop.java:61)
                                                     at no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat.startScan(BluetoothLeScannerCompat.java:112)
                                                     at no.nordicsemi.android.blinky.MainActivity.startLeScan(MainActivity.java:377)
                                                     at no.nordicsemi.android.blinky.MainActivity.onActivityResult(MainActivity.java:345)
                                                     at android.app.Activity.dispatchActivityResult(Activity.java:6808)
                                                     at android.app.ActivityThread.deliverResults(ActivityThread.java:4683)
                                                     at android.app.ActivityThread.handleSendResult(ActivityThread.java:4730) 
                                                     at android.app.ActivityThread.access$1500(ActivityThread.java:197) 
                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1705) 
                                                     at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                     at android.os.Looper.loop(Looper.java:145) 
                                                     at android.app.ActivityThread.main(ActivityThread.java:6873) 
                                                     at java.lang.reflect.Method.invoke(Native Method) 
                                                     at java.lang.reflect.Method.invoke(Method.java:372) 
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404) 
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199) ` 

when i try to turn on/off BLE while the app is running.. I have logged this: 

`03-14 09:47:44.121 21164-21178/no.nordicsemi.android.nrfblinky D/BluetoothAdapter: Bluetooth is turned off, stop adv and scan
03-14 09:47:44.121 21164-21178/no.nordicsemi.android.nrfblinky D/BluetoothLeScanner: stopAllScan standalone boolean is value is = false
03-14 09:47:44.121 21164-21178/no.nordicsemi.android.nrfblinky D/BluetoothLeScanner: Exiting stopAllScan
03-14 09:47:44.751 21164-21219/no.nordicsemi.android.nrfblinky D/BluetoothAdapter: onBluetoothStateChange: up=true
03-14 09:47:44.751 21164-21219/no.nordicsemi.android.nrfblinky D/BluetoothAdapter: onBluetoothStateChange: Bluetooth is on
03-14 09:47:45.951 21164-21164/no.nordicsemi.android.nrfblinky D/ViewRootImpl: ViewPostImeInputStage ACTION_DOWN
03-14 09:47:46.051 21164-21164/no.nordicsemi.android.nrfblinky I/InjectionManager: dispatchOptionsItemSelected :no.nordicsemi.android.blinky.MainActivity
03-14 09:47:46.101 21164-21219/no.nordicsemi.android.nrfblinky D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6` 

Regarding your question to the sony xperia z4 compact all function are workling fine.. scanning, connect to the device, recieving notification from the nRF52,... 

Any idea why the app is crashing, when BLE is turned off? 

Regarding the scanning problem can you give me help for debugging? e.g. where i should place breakpoints? 

Regards Stefan

roshanrajaratnam, 2016-03-14:
Hi Stefan,

I'm beginning to think that this is particularly an issue related to galaxy S5 neo it self. I have tested the app with several other devices including a Samsung Galaxy S6. The app does not crash on my end. Basically it's written in a way to prompt the user to enable Bluetooth if it is turned off. This could be a starting point for your case . You can place break points on the following line numbers. prepareForScan() line number 142 , inside onStart() line number 170 and finally in onActivityResult() on line number 342.

Also if you want to debug the scanning you may put up a couple of breakpoints in the onBatchScanResults inside ScanCallBack. Please note that there is a scanfilter set so that it filters the devices and returns the blinky device/dev kit.

I will try to see if I can find a S5 neo to see if I have the same issues.Happy debugging.

schoensn, 2016-03-15:
Hi Roshan, 

I have start with debugging today .. Unfortunally today I have not been able reproduce the crash when ble is turned off starting the application... so I focused on the scanning issue.

I get `SCAN_FAILED_FEATURE_UNSUPPORTED` in  Line 141 in BluetoothLeScanner.java .. 

If I understand it correctly some settings for scanning are not supported by my device.... 

I have some additonal question.. I ve seen that the compat ble scanner libary is used for scanning in this application.. is this libary also used in the master control panel? I am asking, because scanning works fine with the MCP... 

but I think the MCP does not use a scanfilter? Can you give me hints how to disable the scanfilter to check if this is the reason why scanning does not work?

roshanrajaratnam, 2016-03-15:
Hi Stefan,
Well looking at the scanning failure I assume the same but I cannot come to a conclusion without testing on a device itself. with the MCP there is no library used with the existing source code. Also MCP may successfully scan because the MCP does not use any scan filters. I suggest that you should try removing the scan filter. In the startLeScan method, you can pass a null instead of scanFilterList in `mScanner.startScan(scanFilterList, settings, scanCallback);` on line number 377.

schoensn, 2016-03-15:
Hi,
 to disable the scanFilterList does not change the behavior of the app...

when i disabled the filter I have seen that code line 376 has been commeted out in MainActivity.java .. 

after I activated the code: `.setUseHardwareBatchingIfSupported(false)`

scanning is working fine..also the scanfilter works. 

can you give me information whats the usage of this code line or what exactly hardwarebatching is?

roshanrajaratnam, 2016-03-15:
I think this is specifically related to your device. The reason to set this to false as it is mentioned in the comment (you can find the documentation for this method in the ScanSettings class), some devices have issues when it comes to reporting scan results in batches. if you noticed the scancallback reports the scan results in batches every second instead of scanning through out. Well spotted btw!

closed 2016-03-16 by roshanrajaratnam

schoensn, 2016-03-16:
yes. thanks for your support. :+1:

roshanrajaratnam, 2016-03-16:
Happy to help! if you have any other issues don't hesitate to contact. It would be good to hear how different devices behave.