nordicsemi/Android-nRF-Blinky #4: Error:Unable to resolve dependency for ':app@debugAndroidTest/compileClasspath': Could not resolve project :ble. <a href="openFile:D:/Smart Plud Add on/DemoApplication/Example1/Android-nRF-Blinky-master/app/build.gradle">Open File</a><br><a href="Unable to resolve dependency for &#39;:app@debugAndroidTest/compileClasspath&#39;: Could not resolve project :ble.">Show Details</a>

VinayTShetty, 2018-01-06:
[image]
[image]

[image]
[image]

[image]
**Its showing few errors after opening in android Studio.Can u guide on Resolving it.**
[image]

philips77, 2018-01-08:
Try pulling latest changes in BLE library project and syncing again. I added the ble.iml to github there so importing the library first as a standalone project should not be required anymore.

rajnishkumar9250, 2018-05-06:
@philips77 : I just clone it, giving same error. Please check once again

rajnishkumar9250, 2018-05-06:
@philips77  as follow step 1, there is no build option for this library project.  I think because it is iml file only. I followed step 2 as well. Its correct what u said . still i m facing the same issue.

dohoangminhquan, 2018-05-09:
I have the same issue. Do we have a solution for this ? I can compile BLE.

philips77, 2018-05-09:
Check if the path to the BLE module in settings.gradle file is correct.

dohoangminhquan, 2018-05-09:
All is correct...

dohoangminhquan, 2018-05-09:
It happens with both DFU and BLE

philips77, 2018-05-09:
DFU lib can be added with jcenter dependency.
Could you share what error you get now with nRF Blinky and BLE Library?

dohoangminhquan, 2018-05-09:
in my settings:

include ':app', ':smsmms', ':ExpandableLayout',':libVerrooCore',':libVerrooAndroidBle',':ble',':dfu'

project (':libVerrooCore').projectDir = new File('../rmd-libVerrooCoreJava')
project (':libVerrooAndroidBle').projectDir = new File('../rmd-libVerrooAndroidBle')
project (':ble').projectDir = new File('../Android-BLE-Library')
project (':dfu').projectDir = new File('../Android-DFU-Library')

in my build:
    implementation project(':smsmms')
    implementation project(':ExpandableLayout')
    implementation project(':libVerrooCore')
    implementation project(':libVerrooAndroidBle')
    implementation project(':ble')
    implementation project(':dfu')

And the error I have:

Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve project :ble.
(many errors like this for both dfu and ble)

Only ble and dfu have trouble. other modules are fine.

philips77, 2018-05-09:
You have path to the project, not to the module.
"/ble" and "/dfu" is missing in the paths.
Check out this https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/master/settings.gradle

dohoangminhquan, 2018-05-09:
Thanks, I checked the guide and it works.

philips77, 2018-05-10:
Could you paste a screenshot of your projects folder hierarchy?

philips77, 2018-05-23:
Since version 2.1.0 the app is now using jcenter to import the BLE library dependency.
See bb41ddbe5e0277f16b89a28b46e5b94132ad3a75.

closed 2018-05-23 by philips77

VinayTShetty, 2018-05-23:
> Please, follow those steps:
>
>    1. Open the Android BLE Library project in Android Studio. Check if it
>    builds.
>    2. Open nRF Blinky project in another instance of AS. Check if the
>    Library project name matches one set in gradle settings file in nRF Blinky.
>    By default it should be located in the same root folder as nRF Blinky, with
>    a name 'Android-BLE-Library', as farI remember.

Can please explain via some screenshots.
And post it on issues..

So everyone will be able to resolve it and solve the issue..

Actually what your saying it's confusing.

philips77, 2018-05-23:
You no longer need the Android BLE Library project. Pull recent changes on nRF Blinky project and build it. The BleManager is now imported as a dependency in build.gradle file. No more issues and unslept nights.

philips77, 2018-11-26:
Hi, sorry, but we are not using firebase and have no experience with it. Also, your issue is not related to nRF Blinky. Please, ask on firebase forums, I'm sure someone has already solve it.