nordicsemi/Android-nRF-Blinky #5: No original dex files error

wikpunGitHub, 2018-04-26:
Hi, I got the below error when I run the app. Would you please help?

Suppressed: java.io.IOException: No original dex files found for dex location (arm64) /data/app/no.nordicsemi.android.nrfblinky-a

roshanrajaratnam, 2018-04-26:
Hi @wikpunGitHub have you tried clean building the app.. etc?

wikpunGitHub, 2018-04-26:
Hi Roshan:

Yes, I pressed the "Clean Project" in the "Build" menu inside Android
Studio and it still doesn't work. Here is the complete debug output for
your reference.

*Build Output:*
completed successfully 13s 469ms
Run build 13s 368ms
Load build 5ms
Configure build 84ms
Calculate task graph 302ms
Run tasks 12s 975ms
*Debug Output:*

04/26 13:51:19: Launching app
$ adb install-multiple -r -t
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\resources\instant-run\debug\resources-debug.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\dep\dependencies.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_3.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_2.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_4.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_0.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_5.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_1.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_6.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_7.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_9.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\split-apk\debug\slices\slice_8.apk
C:\Users\Wik\Downloads\Android-nRF-Blinky\app\build\intermediates\instant-run-apk\debug\app-debug.apk
Split APKs installed
$ adb shell am start -n
"no.nordicsemi.android.nrfblinky/no.nordicsemi.android.blinky.SplashScreenActivity"
-a android.intent.action.MAIN -c android.intent.category.LAUNCHER -D
Waiting for application to come online: no.nordicsemi.android.nrfblinky |
no.nordicsemi.android.nrfblinky.test
Waiting for application to come online: no.nordicsemi.android.nrfblinky |
no.nordicsemi.android.nrfblinky.test
Connecting to no.nordicsemi.android.nrfblinky
Connected to the target VM, address: 'localhost:8600', transport: 'socket'
Capturing and displaying logcat messages from application. This behavior
can be disabled in the "Logcat output" section of the "Debugger" settings
page.
E/Zygote: isWhitelistProcess - Process is Whitelisted
E/libpersona: scanKnoxPersonas
              Couldn't open the File - /data/system/users/0/personalist.xml
- No such file or directory
W/SELinux: SELinux selinux_android_compute_policy_index : Policy Index[2],
Con:u:r:zygote:s0 RAM:SEPF_SM-G9650_8.0.0_0002, [-1 -1 -1 -1 0 1]
I/SELinux: SELinux: seapp_context_lookup: seinfo=untrusted,
level=s0:c512,c768, pkgname=no.nordicsemi.android.nrfblinky
I/zygote64: Late-enabling -Xcheck:jni
D/TimaKeyStoreProvider: TimaKeyStore is not enabled: cannot add
TimaSignature Service and generateKeyPair Service
D/ActivityThread: Added TimaKeyStore provider
W/ActivityThread: Application no.nordicsemi.android.nrfblinky is waiting
for the debugger on port 8100...
I/System.out: Sending WAIT chunk
I/zygote64: Debugger is active
I/System.out: Debugger has connected
              waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: debugger has settled (1305)
I/InstantRun: starting instant run server: is main process
D/AndroidRuntime: Shutting down VM
[... 104 lines cut ...]
                      at
dalvik.system.BaseDexClassLoader.<init>(BaseDexClassLoader.java:65)
                      at
dalvik.system.PathClassLoader.<init>(PathClassLoader.java:64)
                      at
com.android.internal.os.PathClassLoaderFactory.createClassLoader(PathClassLoaderFactory.java:43)
                      at
android.app.ApplicationLoaders.getClassLoader(ApplicationLoaders.java:69)
                      at
android.app.ApplicationLoaders.getClassLoader(ApplicationLoaders.java:36)
                      at
android.app.LoadedApk.createOrUpdateClassLoaderLocked(LoadedApk.java:676)
                      at
android.app.LoadedApk.getClassLoader(LoadedApk.java:709)
                      at
android.app.LoadedApk.getResources(LoadedApk.java:942)
                      at
android.app.ContextImpl.createAppContext(ContextImpl.java:2303)
                      at
android.app.ActivityThread.handleBindApplication(ActivityThread.java:5937)
                      ... 8 more

Best Regards,

Wik

wikpunGitHub, 2018-04-26:
Hi Roshan:

I also cleaned it in Gradle pane and it still shows the same error. Please
see the below picture for reference. Thanks!

Best Regards,

Wik

2018-04-26 13:52 GMT+08:00 Wik Pun <wikpun@gmail.com>:

clicit, 2018-07-03:
@wikpunGitHub Try to disable Instant Run!

pardis-sargarmi, 2018-07-17:
kk