nordicsemi/Android-nRF-Blinky #7: Supported bluetooth 5 (coded phy)

devabacus, 2018-05-27:
Hello! It is actually not issue.. but please help me. Could you have any info about term of creating bluetooth 5 long range (coded phy) supported by your ble library. Are there smarthones supported that future nowadays or near future?

philips77, 2018-05-28:
Hi,
I'm working on adding new features to the BLE Library, including requesting phy change.
Check out the lib on develop branch (there's a lot of changes, but no phy support yet).
As far as I know there aren't any phones with coded phy yet. Perhaps Samsung s9, as it's very new,. But probably not. Correct me if I'm wrong.

kostya29, 2018-11-13:
One Plus 6 supports bluetooth 5 long range

philips77, 2018-11-14:
Wow! Could you take a screenshot of the *Device information* screen in nRF Connect and paste here, please?

kostya29, 2018-11-14:
https://photos.app.goo.gl/aMPdNGKQE1HXa7sW6

kb9nate, 2018-12-30:
I can confirm the OnePlus 6T supports the coded phy. I brought a Silicon Labs WSTK with BGM13P into a local T-Mobile store and they let me install nRF Connect on the phone. After that I confirmed I was able to see the coded phy beacons from the WSTK. 

I wonder if it is a matter of the OnePlus 6T having Android Pie vs Android Oreo that some of the other phones have on the market. From what I understand the Bluetooth resides inside the Snapdragon 845 processor, which is in a lot of the new phones out there including the Samsung S9 and Note 9. I had also went to a local store and installed nRF Connect on a Note 9 and it only supported the 2M phy for Bluetooth 5. Maybe a later release will enable it on those phones as well.