nordicsemi/Android-nRF-Blinky #8: problem with Re-establishment of Bonding

arashv6, 2018-05-31:
I am using nrf-blinky for connecting my phone to device with pairing requirements.  For the first time the device advertise without white-list filter and blinky on my phone can bond with it. but for next time, since the device add phone to white list, blinky could not find the device and also can not Re-establishment of Bonding again.  
I can find the bonding device in bonded section of phone system pop-up. but blinky can not find and connect with it. what is the solution?

philips77, 2018-05-31:
Hi, The Blinky sample is very simple. There is no bonding there, that's why we don't support it there. Indeed, after device is bonded, it usually advertises directly with an empty package, so the filter on Blinky fails to find it. Also, newer phones may now show direct adv at all.
The solution would be to obtain a list of bonded devices from the Bluetooth Manager, and connect to it without scanning.
Check out the scanner fragment in Android nRF Toolbox app. There we list also bonded devices on the scanner pop-up. Similar implementation could be added to your app.

arashv6, 2018-05-31:
Thanks for your reply, how about for first time of bonding? is it necessary to change in the source file of Blinky sample? it seems that it can support bonding. in my try some times it can bond but some times after bonding the screen freeze in discovering mode... . do you have any advice in this situation?

arashv6, 2018-06-01:
relating to last issue, sometimes when i try for the first time bond to ble device, the screen get freezing in "Discovering Services..." until i press disconnect button on ble device.  i have attached the logger output. hope to find error.
----------
nRF Blinky, 2018-06-01
SlaveSec_A1 (02:80:E1:94:12:A1)
V 16:12:14.250 Connecting...
D 16:12:14.335 gatt = device.connectGatt(autoConnect = false)
D 16:12:15.043 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D 16:12:15.275 [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
D 16:12:15.359 [Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_PIN (0)
I 16:12:15.405 Connected to 02:80:E1:94:12:A1
D 16:12:27.926 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
I 16:12:27.969 Device bonded
D 16:13:28.516 [Callback] Connection state changed with status: 19 and new state: 0 (DISCONNECTED)
W 16:13:28.693 Error: (0x13): GATT CONN TERMINATE PEER USER
W 16:13:28.759 Connection lost
V 16:13:28.805 Connecting...
D 16:13:28.880 gatt.connect()

philips77, 2018-06-04:
Hi, dis you pull all recent changes? In bonding status broadcasts receiver I've added an extra service discovery. 
You may also try BLE Library on 'develop' branch (that's ver 2 od the library). Do you have a hex file that i could use to test your issue? So I could replicate it on my side.

philips77, 2018-06-04:
I'm taking about this https://github.com/NordicSemiconductor/Android-BLE-Library/commit/7e7d3e4157022e4a620084602cd4fb3b274043b6 commit.

arashv6, 2018-06-04:
thanks philips, I will test your comments. regarding to use version 2 of BLE library, do you think that blinky project can be  compiled with version 2 of BLE libs?

philips77, 2018-06-04:
Requires some refactoring here and there, it will not be it off the box. But I don't think it will fix your issue. Is like to look into it tomorrow.

philips77, 2018-11-06:
Hi, I'm trying to clean up issues on GitHub. The blinky app on *develop* branch is using the BLE Library 2.0-beta now.

closed 2018-11-06 by philips77