nordicsemi/Android-nRF-Blinky #11: BlinkyManager onCharacteristicRead() not getting executed

VinayTShetty, 2018-07-31:
Hi i have changed the UUID in BlinkyManager class, and  i am able to write the command to BLE device sucessfully. And get the Data changed in onCharacteristicNotified.

But the issue is When i get connected to the BLE Device for the Frist time, that time my BLE device will send the status of it , which will be receieved on onCharacteristicRead().But the onCharacteristicRead() is not getting executed.

philips77, 2018-07-31:
Are you reading the status like the original nRF Blinky? In initGatt? 
Did you try the app from develop branch? It's using the new BLE Library version. It won't help, but we'll soon deprecate the older version and develop will become master.

VinayTShetty, 2018-07-31:
yes i am reading the status in initGatt, but onCharacteristicRead() is not getting the call back.And i am not getting what need to be changed in this Blinky code to get the call backs..The callback s i am getting from my BLE device  will be in String formatt . i have not tried the Developer branch..

philips77, 2018-07-31:
Does your device have Read property on the characteristic that you're reading?

VinayTShetty, 2018-07-31:
[image]
[image]
`//	My Custom applicaiton UUID  configuration.

	public final static UUID NORDIC_UART_SERVICE = UUID.fromString("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
    public final static UUID TX_CHARCTERSTIC = UUID.fromString("6e400003-b5a3-f393-e0a9-e50e24dcca9e");
    public final static UUID RX_CHARCTERSTIC = UUID.fromString("6e400002-b5a3-f393-e0a9-e50e24dcca9e");

 //   The UUID name which i have Modified according to Blinky by seeing nrf connect.

	public final static UUID LBS_UUID_SERVICE = UUID.fromString("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
	public final static UUID LBS_UUID_BUTTON_CHAR = UUID.fromString("6e400003-b5a3-f393-e0a9-e50e24dcca9e");
	public final static UUID LBS_UUID_LED_CHAR = UUID.fromString("6e400002-b5a3-f393-e0a9-e50e24dcca9e");
`
### My custom application code which reads the data immediately when its connected to BLE device.
`@Override
        public void onServicesDiscovered(BluetoothGatt gatt, int status) {
            if (status == BluetoothGatt.GATT_SUCCESS) {

                final List<BluetoothGattService> services = gatt.getServices();

                for (BluetoothGattService service : services) {
                    System.out.println("service hrs discoverd " + service.getUuid());

                    if (service.getUuid().equals(NORDIC_UART_SERVICE)) {
                        System.out.println("gap service dicovered" + NORDIC_UART_SERVICE);
                        mLightCharacteristic = service.getCharacteristic(RX_CHARCTERSTIC);
//                        Log.d(TAG, "onServicesDiscovered: " + mLightCharacteristic.getUuid());
                    }

                    if (service.getUuid().equals(NORDIC_UART_SERVICE)) {
                        System.out.println("gap service dicovered");
                        mDeviceNameCharacteristic = service.getCharacteristic(RX_CHARCTERSTIC);
                    }

                    if (service.getUuid().equals(NORDIC_UART_SERVICE)) {
                        mLightCharacteristicNotify = service.getCharacteristic(TX_CHARCTERSTIC);
                        //   mLightCharacteristicRead = service.getCharacteristic(LIGHT_CHARACTERISTIC_READ_UUID);
                        setCharacteristicNotification(mLightCharacteristicNotify, true);
                        System.out.println("service discoverd ");
                    }

                }
             //   broadcastUpdate(ACTION_GATT_SERVICES_DISCOVERED);
            } else {
                Log.w(TAG, "onServicesDiscovered received: " + status);
            }
        }`

VinayTShetty, 2018-07-31:
**Yes my device has read property.**But in Blinky applicaiton code i am not getting where i can read the charcterstics.

VinayTShetty, 2018-07-31:
Can u please guide me on how to read the data immediately when the device gets connected to the application. 

now i am able to read the data only when i write the data in (onCharacteristicWrite()), it notifies  me on Oncharctersticnotified(onCharacteristicNotified()).

But i want to read the data when the Blinky application get connected to the device.

philips77, 2018-07-31:
Are you using UART service? None of those characteristics have Read property, which is shown on the screenshot you posted. There's no Read button there, like one next to both Button and LED characteristics.

VinayTShetty, 2018-07-31:
Hi @philips77  thanks for the instant response.In nrf connect i am sending smpon on RX charcterstic and i am instantly getting ON in TX charcterstic. as shown in the screen shot..
[image]
[image]

philips77, 2018-07-31:
Yes, but that's not Reading.
In nRF Blinky you'll never get onChar..Read, but rather onChar...Notified.

VinayTShetty, 2018-07-31:
ok thanks. suppose assume if the nrf blinky BLE device is in ON state previously, i open the application immediately the LED state will get into ON state in the application.how to achieve this if we are customizing the UUID...?
 Querying the BLE device during starting of the application and Modifying according to it.

philips77, 2018-07-31:
In initGatt you'd have to first enable notifications, then write a command to get the state and you'll receive a notification with this state.

VinayTShetty, 2018-07-31:
Can u please tell me where i need to modify  the enable notification to write a command to get the state in which classes and methods i need to modify.it would be really helpfull.

philips77, 2018-07-31:
Looks like this is something you want to do after you connect to this device, so you have to modify initGatt method. In nRF Blinky there are 3 requests added to the queue: read, read and enable notifications. Check BlinkyManager class.
You have to remove the 2 reads and add one/two writes after enabling notifications.
For each write you should get a onChar..Notified callback, if I understand correctly, with the values you wanted.

VinayTShetty, 2018-07-31:
[image]
Here which write request to be used and what arguments to be passed can u guide me please

VinayTShetty, 2018-07-31:
[image]
Its not working as i write after enable notification.please let me know any other alternatives.as there is no 
onChar..Notified called when the application is connected to device

philips77, 2018-07-31:
Well... I have no idea how your device works and what your have to write to get some status.
You wrote before, that you should be getting a notification after a write. It's up to you now.

VinayTShetty, 2018-08-07:
Hi @philips77 thank you for the time.can u please tell in brief what charcterstics i need to write it. seeing the Screen Shot that i have posted.
As i am stuck in problem for couple of days

VinayTShetty, 2018-08-12:
@philips77 thanks for the tip i am writing in the que in the BlinkyManager class which i will get notification on Charcterstic Read ater connection immediately thanks a  ton.It works.

closed 2018-08-12 by philips77