nordicsemi/Android-nRF-Blinky #18: Latest version is not working correctly.

devangichhatbar, 2018-11-15:
**Scanner screen is continuously scanning. But did not able to scan device.** 

Here I am attaching screenshot. I can't able to move from this screen.
[image]

philips77, 2018-11-15:
HI, I've just tested the app from Play Store and it works for me. Do you see your device using nRF Connect? Do you have Location enabled?

devangichhatbar, 2018-11-15:
Yes. I see my device using nRF connect. Yes, Location is enabled.

philips77, 2018-11-15:
Is it advertising with `00001523-1212-efde-1523-785feabcd123` service UUID?

devangichhatbar, 2018-11-15:
No, I am using my own device service UUID.

philips77, 2018-11-15:
That's why it's not visible. nRF Blinky filters only for this UUID. You may clone the project from GitHub and modify to match your needs.

closed 2018-11-15 by philips77

philips77, 2018-11-15:
The Play Store version is expecting the UUIDs provided above in device's advertising packets. You may try emulating Blinky sample with nRF Connect on another phone and adding the Blinky Service UUID into the advertisement.

Pareshoct7, 2018-11-22:
Yes, Similar kind of issue. But I just install app from play store and then try to connect. And It is perfectly working with nRF connect. But not with nRF Blinky.

philips77, 2018-11-22:
Do you see your device in nRF Blinky? Check in nRF Connec if it is advertising with Blinky UUID. The nRF Blinky app filters only this UUID.

Pareshoct7, 2018-11-22:
How can I customize with my device? Where I have to change the UUID?

philips77, 2018-11-22:
Hi, I thought it will be easiest when I release the new version with scanner filters. Now, using the menu, you may deselect requirement for UUID and all devices in range will be shown. However, Blinky Service ( aka LBS, Led-Button Service) is still required after connection. The new version, 2.4.0, is available now on GitHub and in few moments on Google Play.

philips77, 2018-11-23:
Hi, did the new version fix your problem?