nordicsemi/Android-nRF-Blinky #20: Failed to build v2.4.0 on Android Studio 3.3.0

tddpirate, 2019-01-20:
Android Studio 3.3.0
Project build gradle: classpath 'com.android.tools.build:gradle:3.3.0'
gradle/wrapper/gradle-wrapper.properties: distributionUrl=https\://services.gradle.org/distributions/gradle-4.10.1-all.zip

When trying to build the project:
1.
> Task :app:generateDebugSources UP-TO-DATE
>
> Transform error_prone_annotations.jar (com.google.errorprone:error_prone_annotations:2.1.3) with JetifyTransform
> ERROR: [TAG] Failed to resolve variable '${junit.version}'

2.
> Task :app:validateSigningDebug
>
> Transform guava.jar (com.google.guava:guava:24.0-jre) with JetifyTransform
> ERROR: [TAG] Failed to resolve variable '${animal.sniffer.version}'

3.
> Task :app:javaPreCompileDebug FAILED
>
> FAILURE: Build failed with an exception.
>
> * What went wrong:
> Could not resolve all files for configuration ':app:debugAnnotationProcessorClasspath'.
> Failed to transform file 'butterknife-compiler-9.0.0-rc1.jar' to match attributes {artifactType=processed-jar} using transform JetifyTransform
> Failed to transform '/AndroidStudio/home/.gradle/caches/modules-2/files-2.1/com.jakewharton/butterknife-compiler/9.0.0-rc1/e66446c78b80f3e91fad9d222942aa3b7d26355/butterknife-compiler-9.0.0-rc1.jar' using Jetifier. Reason: The given artifact contains a string literal with a package reference 'android.support.v4.content' that cannot be safely rewritten. Libraries using reflection such as annotation processors need to be updated manually to add support for androidx.. (Run with --stacktrace for more details.)

philips77, 2019-01-30:
Modify the butterknife dependency to 10.0.0.

philips77, 2019-02-07:
Try now with the latest changes on *master* or *develop* branches.

closed 2019-02-07 by philips77