nordicsemi/Android-nRF-Blinky #22: A device that stops advertising is not removed from the UI

eliotstock, 2019-02-17:
Hi. I've used this code as a starting point for an activity that scans for a device and allows the user to connect to one of the found devices. I've found that, when my device stops advertising, the view for it in the list does not disappear. It's therefore easy to attempt connecting to a device to which the user can no longer connect, after which there's a timeout and error 133.

If the Blinky never stops advertising (and I don't know whether it does or not), this is less of an issue for this codebase. But given this is a starter project for apps like mine, I think it makes sense to implement this.

philips77, 2019-02-18:
Hi again Eliot,
The ux of removing a row from device list is not good. It may cause an unwanted click, if a user tries to click a device but it disappears at the same moment and a different one is clicked. But the UI could of course indicate it in a different way, by hiding the RSSI indicator or making text gray. I'll add it to our backlog, but with a low priority, as it's full with other tasks already.
Thanks for the suggestion!

eliotstock, 2019-02-18:
It's a good point about the "tap and miss" problem. That would need to be balanced against the UX when a user taps a device that's no longer advertising or no longer in range, though. What is that UX right now in the Blinky app? A 30s wait, error 133 and then some sort of error message to the user? That's arguably worse than the small chance of a tap and miss.

FWIW, there is precedent in Android for lists of things that reorder and from which items disappear: the WiFi AP picker does this. I guess one way to allow it and still prevent a tap and miss would be to disconnect the click handler from all items a second before changing the list, then reattach it a second after. A poorly timed tap would do nothing. It's a bit of extra work and complexity though.

eliotstock, 2019-02-20:
I just tested the WiFi AP picker in the phone setup on a Pixel and this is exactly what they do: right before changing the list, all list items are disabled for a short period, then the list is modified, then they're re-enabled.

istvanns, 2019-04-23:
I also used this app as the basis for my own. I just added a refresh button to the options menu of the scanner activity and execute `clear()` when it's clicked. It refreshes the entire list, requires minimal work to implement. By the way, the scanner part of this app is very useful, thanks to Nordic for open sourcing it!

minghuadev, 2023-05-07:
In version 3.0.1, holding and pulling the page down, will do a refresh. The list will be updated by that. But that does not always reliably work.

I also noticed, if you click the "all" button and click "all" again to go back, the scanner will miss updating the scanned device which still shows in the list. Then click "nearby" and click it again, the scanner will continue to update the shown device again. Likely a bug in the scanner which is in the common library repository but that repository does not have an issue tab on github. Maybe that can be fixed by calling `clear()` somewhere too?