nordicsemi/Android-nRF-Blinky #24: Calling stopScan on a device were the BT adapter is not turned ON causes an exception

RobinCaroff, 2019-05-23:
I ran the sample with Bluetooth turned off then pressed back on the ScannerActivity. It causes a crash with the following stack trace :

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: no.nordicsemi.android.nrfblinky, PID: 5260
    java.lang.RuntimeException: Unable to stop activity {no.nordicsemi.android.nrfblinky/no.nordicsemi.android.blinky.ScannerActivity}: java.lang.IllegalStateException: BT Adapter is not turned ON
        at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:5117)
        at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:5174)
        at android.app.ActivityThread.-wrap6(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2058)
        at android.os.Handler.dispatchMessage(Handler.java:108)
        at android.os.Looper.loop(Looper.java:166)
        at android.app.ActivityThread.main(ActivityThread.java:7529)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:245)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:921)
     Caused by: java.lang.IllegalStateException: BT Adapter is not turned ON
        at no.nordicsemi.android.support.v18.scanner.r.a(:140)
        at no.nordicsemi.android.support.v18.scanner.o.b(:90)
        at no.nordicsemi.android.support.v18.scanner.d.a(:205)
        at no.nordicsemi.android.blinky.viewmodels.ScannerViewModel.stopScan(:164)
        at no.nordicsemi.android.blinky.ScannerActivity.stopScan(:225)
        at no.nordicsemi.android.blinky.ScannerActivity.onStop(:102)
        at android.app.Instrumentation.callActivityOnStop(Instrumentation.java:1386)
        at android.app.Activity.performStop(Activity.java:7587)
        at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:5112)
        at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:5174) 
        at android.app.ActivityThread.-wrap6(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2058) 
        at android.os.Handler.dispatchMessage(Handler.java:108) 
        at android.os.Looper.loop(Looper.java:166) 
        at android.app.ActivityThread.main(ActivityThread.java:7529) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:245) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:921) 
```

Bluetooth adapter state or mScannerStateLiveData should be checked before trying to stop the scanner.

philips77, 2019-05-23:
Thanks for the bug report. We will look into it when we have some time.

RobinCaroff, 2019-05-23:
Cool. I sent a PR for this issue.

roshanrajaratnam, 2020-04-08:
@RobinCaroff thanks for your contribution.

closed 2020-04-08 by roshanrajaratnam