nordicsemi/Android-nRF-Blinky #26: How to fix switch LED?

ArTyGill, 2019-05-24:
In the latest version of Blinky, the LED turns on, but does not allow it to be turned off.

ArTyGill, 2019-05-24:
Solved the problem by deleting 188 and 189 lines in the file "BlinkyManager.java" .

https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/master/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java

ArTyGill, 2019-05-27:
How can I fix the problem on the controller so that the application works out of the box?

philips77, 2019-05-27:
Doesn't it work out of the box with the unmodified Blinky sample from the sdk?

ArTyGill, 2019-05-28:
I use esp32 as a controller and I wrote the firmware on it myself, everything was fine, but with the latest version of the Nrf Blinky I can turn on and can not turn off the LED. Without correction in the code of one line.

ArTyGill, 2019-05-28:
I get a warning "Invalid data received: ", possible with this problem. And it prevents me from receiving a callback  onLedStateChanged.
What is the callback onLedStateChanged? What is this callback for?

ArTyGill, 2019-05-28:
Now everything is rebuilt and it seems to work. Sources work.

closed 2019-05-28 by ArTyGill