nordicsemi/Android-nRF-Blinky #28: App Crashes

khan-07, 2019-07-26:
Hi,

I am trying to run this blinky program on an huawei nova 3i. The app crashes as soon as it is launched. My bluetooth is on as well as i have given locations permissions to the device. I am building the project using android studio. What can be the possible causes ?

`Process: no.nordicsemi.android.nrfblinky, PID: 14788
    java.lang.BootstrapMethodError: Exception from call site #8 bootstrap method
        at no.nordicsemi.android.blinky.SplashScreenActivity.onCreate`

It fails at the when the handler is being called. What can be the possible cause of it ?

philips77, 2019-07-28:
Why there's bootstrap involved?

roshanrajaratnam, 2020-04-08:
@khan-07 Closing due to inactivity, please feel free to re-open this if the issue persists.

closed 2020-04-08 by roshanrajaratnam