nordicsemi/Android-nRF-Blinky #31: Any change in the ScannerStateLiveData will restart the scan

eliotstock, 2019-10-25:
Here's how you hook your `ScannerViewModel` up in your activity:

```
// Create view model containing utility methods for scanning
mScannerViewModel = ViewModelProviders.of(this).get(ScannerViewModel.class);
mScannerViewModel.getScannerState().observe(this, this::startScan);
```

Are you sure this is a good idea?

This means that any change in the `ScannerStateLiveData`, including a new scan result showing up, will cause a `postValue()` which will cause another call to `startScan()`. That then calls `ScannerViewModel.startScan()`, which bails out if it's already scanning. So far so good.

The problem comes when I want to stop scanning once I've tapped on a device. `ScannerViewModel.stopScan()` will call `scannerStateLiveData.scanningStopped()`, but that will immediately call `postValue()` which will cause the activity's `startScan()` method to be called again. Given I'm in the middle of trying to connect to a device at this point, I think this is bad for the Android BLE stack.

philips77, 2019-10-27:
Thank for reporting this. We have no time to fix it soon, but will eventually handle it.

philips77, 2025-01-09:
The app has been rewritten, and this issue is obsolete now.

closed 2025-01-09 by philips77