nordicsemi/Android-nRF-Blinky #32: android 10 with a pixel 3a

arrow53, 2019-11-29:
I can't seem to get the blinky peripheral to show up when I build this app or run it out of google play store. 

I can get the nrf connect app to see it fine and connect/disconnect from it. Under CLIENT I can also see the button/led and interact with them fine.

The ios version of this app works fine.

Any ideas on what is needed to make this work?

arrow53, 2019-11-29:
@philips77 btw, I did see #18. I tried using my ios device broadcasting the nordic led and button service with that uuid. Again, on android the connect app can see it but the nrf blinky app cannot.

philips77, 2019-11-29:
You may need to switch the requested permission from location coarse to location fine. In Android manifest and in the coffee, where is requested in runtime. 
We need to update all our apps but that may take some time.

arrow53, 2019-11-29:
ACCESS_FINE_LOCATION did the trick. thanks! I'm going to assume you're tracking this and go ahead and close.

closed 2019-11-29 by arrow53