nordicsemi/Android-nRF-Blinky #37: Use BlinkyManager Singleton in Viewmodel

zawawimanja, 2020-01-24:
I noticed that you don't use the singleton blinkymanager in ViewModel. Is there any reason?

roshanrajaratnam, 2020-01-24:
Hi, I would suggest creating your repository as a Singleton instead of the BlinkyManager and create your BlinkyManager instance in the repository. Basically move the connection establishment to the repository from the ViewModel. This would let you access the same BlinkyInstance from the BleRepository and observe the connection states from almost any activity.

zawawimanja, 2020-01-24:
Hi , Actually I have  replaced BlinkyManager as BlinkyRepository. Then in
ViewModel I am  calling the singleton BlinkRepository but failed.

Or what you means is to create other file name Repository. Then in
repository , instantiate the BlinkyManager in Repository. Instantiate the
Repository as Singleton and call it in ViewModel ?So I have 2 files for
doing this  BlinkyManager and Repository  compare above method only
requires  1 files only Repository.

roshanrajaratnam, 2020-04-08:
I don't think having two class files here will make major impact. This is more easier to deal with in your case having one BlinkyManager instance in your BlinkyRespository and having all your viewmodels access the same manager that handles the connections. hope this helps.

closed 2020-04-08 by roshanrajaratnam