nordicsemi/Android-nRF-Blinky #38: Scanning fail on Samsung One UI 2.0(android 10)

jd1017, 2020-04-08:
BLE scanning fails in Samsung One UI 2.0.
It is working as expected in other android flavors. Even works with Samsung One UI 1.5.
I have attached images for the same and please ignore BLE name.
[image]
[image]

roshanrajaratnam, 2020-04-08:
@jd1017 From Android 10 onwards "ACCESS_FINE_LOCATION" is required, we may have it not fixed yet in this version. However for now you can go in to app settings disable and enable location permissions. Also make sure location services are enabled too.

jd1017, 2020-04-08:
Thanks for the quick response! i have tried that but didn't worked and also i have downloaded the source code and updated with FINE_LOCATION still i am not getting result.

roshanrajaratnam, 2020-04-08:
@jd1017 Please look in to Runtime permissions in Android as now you will have to request for this permission from the user. You only need ACCESS_FINE_LOCATION for Android 10 and above and ACCESS_COARSE_LOCATION for device below Android 10. Disabling and enabling is a work around for you to get up and running, so make sure you request the user for the required permissions if not the device will not give you any scan results.

jd1017, 2020-04-08:
Updating code to check API level and providing permission accordingly is working in some cases.

closed 2020-04-08 by jd1017