nordicsemi/Android-nRF-Blinky #43: The broadcast time is changed to 2 seconds, often unable to connect successfully

lp292650230, 2020-06-04:
Use SDK16 blinky firmware, modify the broadcast interval to 2 seconds, use Android nRF-blinky connection, often unsuccessful connection phenomenon, APP interface is displayed as Initing.... Use nRF-connect to connect successfully every time. (Mobile system AndroidX).
Modify the broadcast interval to 100 milliseconds, you can connect successfully using Android nRF-blinky
Do I need to modify any parameters of nRF-blinky?

philips77, 2020-06-04:
Could you post the logs from LogCat? it appears that nRF Blinky has sent some request and it's waiting for a callback, but I don't know which request. It's one of those:
https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/2f3a06bf0e0c62a8172ac7fe2d872ac62ec19d84/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java#L158-L163

lp292650230, 2020-06-05:
Sorry，Correct the following，The nRF-blinky  #41   application interface always shows Connecting。
LogCat has been sent to you by e-mail, and please help analyze the reason.

closed 2025-01-09 by philips77