nordicsemi/Android-nRF-Blinky #52: Not able to scan for *only* Coded PHY

mafaneh, 2021-09-14:
I am currently using a OnePlus Nord N10 5G and I can confirm it works well for scanning, advertising (primary & secondary), and connecting all using Coded PHY (using the nRF Connect Android app).

The issue I am running into is I'm not able to scan only Coded PHY advertisements (primary + secondary). According to the Android API documentation, you can set the PHY used for scanning to be: 1M + Coded PHY, 1M only, or Coded only. However, I am still receiving non-Coded PHY advertisements even when I set the Phy to Coded PHY using the .setPhy(BluetoothDevice.PHY_LE_CODED) during the setting of the Scan Settings.

Here's the code I modified in the app (in ScannerViewModel.java):

```java
	public void startScan() {
		if (scannerStateLiveData.isScanning()) {
			return;
		}

		// Scanning settings
		final ScanSettings settings = new ScanSettings.Builder()
				.setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
				.setLegacy(false)
				.setPhy(BluetoothDevice.PHY_LE_CODED)
				.setReportDelay(500)
				.setUseHardwareBatchingIfSupported(false)
				.build();

		final BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
		scanner.startScan(null, settings, scanCallback);
		scannerStateLiveData.scanningStarted();
	}
```

Is this an issue with the API? or am I doing something wrong?

philips77, 2021-09-14:
How about you set report delay to 0?

mafaneh, 2021-09-14:
I have not tried other phones. I only have a Pixel 4a and I believe that does not support scanning for extended advertisements. In your testing, have you tried scanning for Coded PHY only?

philips77, 2021-09-15:
We rather checked if devices advertising with primary phy coded are discovered at all, but we used "all supported" mode for phy no in the scanner.

closed 2025-01-09 by philips77