nordicsemi/Android-nRF-Blinky #67: Disconnects from peripherial (nRF for iOS simulator) with not supported error

mecoFarid, 2022-01-28:
Trying to connect to an nRF for iOS simulator that is installed on iPhone.  But I can't connect in any way. The following are the visualization of the problem in steps with pictures.

 ## Advertiser UI:
 [image]
 [image]

## Android Bluetooth interface. Device is shown in the list
 [image]

 ## The device is listed as totally different name and when I try to connect to it displays the following error:
 [image]
 [image]

philips77, 2025-01-09:
It 2 years now since you asked, so time for a reply ;)
I believe you didn't set up the TX and RX characteristics in the LBS service, hence the app disconnects.

mecoFarid, 2025-01-09:
@philips77 The question is so old, that I don't even remember whether I got it working or not LOL.

Thanks for answer tho 🙌