nordicsemi/Android-nRF-Blinky #79: Process 'command 'git'' finished with non-zero exit value 128

kriskak, 2023-02-13:
Hello,
I downloaded the Repository as .zip, unpacked and tried to import to Android Studio to try it out.
But I get this Error from Gradle sync: Process 'command 'git'' finished with non-zero exit value 128
I do not know what to do. I didn't find the solutions from Stackoverflow and other git issues aplicable for this problem.
What can I do about it?
I am thanking you in advance for your help.

philips77, 2023-02-13:
Our build script expects the repo to be cloned, not download as zip. Please, use git to clone instead for now.
We will fix it soon.

kriskak, 2023-02-15:
ok thank you!

stonehiy, 2023-02-17:
An exception occurred applying plugin request [id: 'no.nordicsemi.android.gradle.application.compose', version: '1.2.0']
> Failed to apply plugin 'no.nordicsemi.android.gradle.application'.
   > Process 'command 'git'' finished with non-zero exit value 128

philips77, 2023-02-21:
Please, modify version catalog version below to 1.3.1:
https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/1e3130ba3c520130359217b9fc2905f7741b2700/settings.gradle.kts#L19

closed 2025-01-09 by philips77