nordicsemi/Android-nRF-Blinky #92: Getting an error while trying to build the app

Gryk3l, 2025-01-01:
Hi,

First of all, sorry if this is a very dumb/basic question, I am pretty much new in the Android dev. world.

I just cloned the repo, opened it with Android Studio and tried to build it. When I do so, I get the following error:

```PS C:\<Personal route>\Android-nRF-Blinky> .\gradlew clean build
Starting a Gradle Daemon, 1 incompatible and 4 stopped Daemons could not be reused, use --status for details

> Task :scanner:kaptGenerateStubsDebugKotlin
w: Support for language version 2.0+ in kapt is in Alpha and must be enabled explicitly. Falling back to 1.9.

> Task :app:stripDebugDebugSymbols
Unable to strip the following libraries, packaging them as they are: libandroidx.graphics.path.so.

> Task :scanner:kaptDebugKotlin                                                                                                                                                                          
[WARN] Issue detected with dagger.internal.codegen.ComponentProcessor. Expected 1 originating source file when generating C:\<Personal route>\Android-nRF-Blinky\scanner\build\generated\source\kapt\debug\no\nordicsemi\android\scanner\main\viewmodel\ScannerViewModel_HiltModules_BindsModule_Binds_LazyMapKey.java, but detected 0: [].

> Task :scanner:kaptDebugKotlin FAILED                                                                                                                                                                   

FAILURE: Build failed with an exception.

* What went wrong:                                                                                                                                                                                       
Execution failed for task ':scanner:kaptDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.gradle.internal.KaptWithoutKotlincTask$KaptExecutionWorkAction

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 1m 8s
124 actionable tasks: 124 executed
```

This is probably an error on my side, but I haven't had much luck looking for a solution.

If there is any extra info you want, please feel free to ask for it.

philips77, 2025-01-08:
Please update version catalog version in _settings.gradle.kt_ file to 2.6.1.

philips77, 2025-01-08:
https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/eb680ba248a4b649bc424acc6046f007fbffc54f/settings.gradle.kts#L19

philips77, 2025-01-08:
This issue is related to Dagger / Hile in version 2.53 and only occurs on Windows, where paths use `\`, but `/` in .jar files.
Dagger was using `\` in both cases. This was fixed in 2.54.

Gryk3l, 2025-01-08:
Hi,

Thanks for the quick response. I've tried with versions 2.5-4 and 2.5-5 and the error persists. When trying with catalog version 2.6 (that would be `from("no.nordicsemi.android.gradle:version-catalog:2.6")`), I get the following `error:`

```
Build file 'C:\Users\<Redacted>\Desktop\Android-nRF-Blinky\scanner\build.gradle.kts' line: 19

Cannot convert the provided notation to an object of type Dependency: org.gradle.accessors.dm.LibrariesForLibs$AndroidxComposeMaterialLibraryAccessors@5f8ea7b4.
The following types/formats are supported:
  - String or CharSequence values, for example 'org.gradle:gradle-core:1.0'.
  - Maps, for example [group: 'org.gradle', name: 'gradle-core', version: '1.0'].
  - FileCollections, for example files('some.jar', 'someOther.jar').
  - Projects, for example project(':some:project:path').
  - ClassPathNotation, for example gradleApi().

* Try:
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

* Exception is:
org.gradle.internal.typeconversion.UnsupportedNotationException: Cannot convert the provided notation to an object of type Dependency: org.gradle.accessors.dm.LibrariesForLibs$AndroidxComposeMaterialLibraryAccessors@5f8ea7b4.
The following types/formats are supported:
  - String or CharSequence values, for example 'org.gradle:gradle-core:1.0'.
  - Maps, for example [group: 'org.gradle', name: 'gradle-core', version: '1.0'].
  - FileCollections, for example files('some.jar', 'someOther.jar').
  - Projects, for example project(':some:project:path').
  - ClassPathNotation, for example gradleApi().

Comprehensive documentation on dependency notations is available in DSL reference for DependencyHandler type.
	at org.gradle.internal.typeconversion.ErrorHandlingNotationParser.parseNotation(ErrorHandlingNotationParser.java:57)
	at org.gradle.api.internal.artifacts.DefaultDependencyFactory.createDependency(DefaultDependencyFactory.java:79)
	at org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.create(DefaultDependencyHandler.java:172)
	at org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.doAddRegularDependency(DefaultDependencyHandler.java:201)
	at org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.doAdd(DefaultDependencyHandler.java:196)
	at org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.add(DefaultDependencyHandler.java:124)
	at org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.add(DefaultDependencyHandler.java:118)
	at org.gradle.kotlin.dsl.support.delegates.DependencyHandlerDelegate.add(DependencyHandlerDelegate.kt:53)
	at org.gradle.kotlin.dsl.ImplementationConfigurationAccessorsKt.implementation(Unknown Source)
	at Build_gradle$2.invoke(build.gradle.kts:19)
	at Build_gradle$2.invoke(build.gradle.kts:12)
	at org.gradle.kotlin.dsl.ProjectExtensionsKt.dependencies(ProjectExtensions.kt:237)
	at Build_gradle.<init>(build.gradle.kts:12)
	at Program.execute(Unknown Source)
	at org.gradle.kotlin.dsl.execution.Interpreter$ProgramHost.eval(Interpreter.kt:514)
	at org.gradle.kotlin.dsl.execution.Interpreter$ProgramHost.evaluateSecondStageOf(Interpreter.kt:426)
	at Program.execute(Unknown Source)
	at org.gradle.kotlin.dsl.execution.Interpreter$ProgramHost.eval(Interpreter.kt:514)
	at org.gradle.kotlin.dsl.execution.Interpreter.eval(Interpreter.kt:193)
	at org.gradle.kotlin.dsl.provider.StandardKotlinScriptEvaluator.evaluate(KotlinScriptEvaluator.kt:124)
	at org.gradle.kotlin.dsl.provider.KotlinScriptPluginFactory$create$1.invoke(KotlinScriptPluginFactory.kt:51)
	at org.gradle.kotlin.dsl.provider.KotlinScriptPluginFactory$create$1.invoke(KotlinScriptPluginFactory.kt:48)
	at org.gradle.kotlin.dsl.provider.KotlinScriptPlugin.apply(KotlinScriptPlugin.kt:35)
	at org.gradle.configuration.BuildOperationScriptPlugin$1.run(BuildOperationScriptPlugin.java:68)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:166)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.run(DefaultBuildOperationRunner.java:47)
	at org.gradle.configuration.BuildOperationScriptPlugin.lambda$apply$0(BuildOperationScriptPlugin.java:65)
	at org.gradle.internal.code.DefaultUserCodeApplicationContext.apply(DefaultUserCodeApplicationContext.java:44)
	at org.gradle.configuration.BuildOperationScriptPlugin.apply(BuildOperationScriptPlugin.java:65)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$applyToMutableState$1(DefaultProjectStateRegistry.java:407)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.fromMutableState(DefaultProjectStateRegistry.java:425)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.applyToMutableState(DefaultProjectStateRegistry.java:406)
[... cut ...]
	at org.gradle.internal.concurrent.AbstractManagedExecutor$1.run(AbstractManagedExecutor.java:48)
```

The line giving the error is:
https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/eb680ba248a4b649bc424acc6046f007fbffc54f/scanner/build.gradle.kts#L19

Feel free to close the issue at any point now, since I was able to grab the parts of the code from the app that I needed, which means I don't need to be able to build the project anymore.

However, I can try any other changes you suggest to try to fix this, in case you think it can be useful for someone else. 

Thanks again!

philips77, 2025-01-08:
OK, I'll fix it asap. Sorry!

PR #93 (2025-01-08): Migration to Material 3
This PR fixes #92.

`material` dependency was removed in AGP 2.6.
closed 2025-01-08 by philips77